KmanOz/Sonoff-HomeAssistant #2: Not an issue more of a question

nbittmann, 2016-11-14:
Do you think it's possible to change the code so that we can add an external switch? 

I'm just thinking of the scenario that I might want to use a switch to control the lifts as well I could just reuse the switch that's on the board but I was wondering if there is any way to use the GPIO 14 which you use for the temp sensor for it?

Thanks

KmanOz, 2016-11-17:
Have a look at the discussion forum here [https://community.home-assistant.io/t/sonoff-homeassistant-alternative-firmware-for-sonoff-switches-for-use-with-mqtt-ha/2332/77](url) You can literally parallel a switch over the current switch if you want external control. Although it would take little to no time to do what you suggesting, it's an extra version I would need to get put there when it's almost not necessary if you do what that person did.

nbittmann, 2016-11-21:
@KmanOz Thanks a lot actually saw that earlier on as well and that will serve my purpose fine as well one more question.

I'm in process of making a remote light switch with the ESP8266 and was thinking to reuse your code since it seems rock solid not clutter in here and it works but I'd just want to use a button and an led can I comment out the 

` pinMode(RELAY, OUTPUT);`

to deactivate action on the relay GPIO pin? 

If yes then I could pretty much just use your code as a drop in and connect the same way as the Sonoff relay button to GPIO 0 and the Led to GPIO 13 if I'm not mistaken.

Thanks a lot and apologies for asking these questions :)

KmanOz, 2016-11-21:
Yep no problems. You pretty much got it. As long as your switch is on GPIO 0 (so you can send code to it) you can pretty much put your LED on any of the GPIO pins. What you have will work fine. In fact I had ESP's doing what you're doing well before Sonoff introduced their switch. I just adapted that code for the Sonoff's and cleaned it up so it was easy to follow for all. So you're good to go. Good luck.

nbittmann, 2016-11-21:
Awesome this makes my live much easier and I can start using it as a very cheap switch :)

Thanks a lot for your help I'll close this out!!

closed 2016-11-21 by nbittmann