KmanOz/Sonoff-HomeAssistant #9: Sonoff Touch

Chariyski, 2017-02-11:
Hi @KmanOz,

I have used the ESPsonoff-v1.01p on my sonoff and it is working perfectly, but on my sonoff touch it is not working at all. 

Have you tested it on a real sonoff touch and is there something different that need to be done for the touch version?

Best regards

KmanOz, 2017-02-12:
No I haven't tested it personally. I don't have one on hand. In the HomeAssistant discussion forum https://community.home-assistant.io/t/sonoff-homeassistant-alternative-firmware-for-sonoff-switches-for-use-with-mqtt-ha/2332 there are a few users who have successfully flashed the unit and it works fine for them hence why I modified the documentation and said it is supported. 

Sorry about that.

Chariyski, 2017-02-12:
Hi @KmanOz, 

I've tested with v1.0, but it is not working again. Maybe I'm doing something wrong. 

This is what I've changed in the configuration:
```

#define MQTT_CLIENT     "Sonoff_Bedroom_Room_v1.0p"          // mqtt client_id (Must be unique for each Sonoff)
#define MQTT_SERVER     "192.168.0.150"                      // mqtt server
#define MQTT_PORT       1883                                 // mqtt port
#define MQTT_TOPIC      "home/sonoff/bedroom/1"              // mqtt topic (Must be unique for each Sonoff)
#define MQTT_USER       "secret"                             // mqtt user
#define MQTT_PASS       "secret"                             // mqtt password

#define WIFI_SSID       "secret"                            // wifi ssid
#define WIFI_PASS       "secret"                            // wifi password
```

Do you see any mistake?

Best regards

KmanOz, 2017-02-13:
There should be no reason why it doesn't work. I have checked the pins they use from the ESP and they're the same in the Touch and Sonoff. What exactly happens (or doesn't). Can you leave serial plugged in and power it via that, do not connect it to mains and tell me what happens in the serial console.

Chariyski, 2017-02-13:
I think that maybe there is something on my side...
The serial monitor prints unknown characters... but can't find a way to fix it.

KmanOz, 2017-02-14:
Is your baud rate set to 115200 in your terminal?

Chariyski, 2017-02-15:
yes, and I uploaded the code with the same speed, too

KmanOz, 2017-02-15:
Are you setting the Arduino IDE to ESP8285?

Chariyski, 2017-02-15:
:flushed: silly me 😄 

Know everything is working. If you want I can add documentation to your repo.

Thanks a lot for the support and the code.

closed 2017-02-15 by Chariyski

KmanOz, 2017-02-15:
Go for it. Enjoy