KmanOz/Sonoff-HomeAssistant #13: compiling error

jonny190, 2017-04-04:
Any Ideas on this compiling issue? 

`Arduino: 1.8.2 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

Build options changed, rebuilding all
ESPsonoff-v1.0p:35: error: no matching function for call to 'PubSubClient::PubSubClient(WiFiClient&, const char [14], int)'

 PubSubClient mqttClient(wifiClient, MQTT_SERVER, MQTT_PORT);

                                                           ^

C:\Users\Jonny\Desktop\Sonoff-HomeAssistant-master\arduino\ESPsonoff-v1.0p\ESPsonoff-v1.0p.ino:35:59: note: candidates are:

In file included from C:\Users\Jonny\Desktop\Sonoff-HomeAssistant-master\arduino\ESPsonoff-v1.0p\ESPsonoff-v1.0p.ino:2:0:

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:116:4: note: PubSubClient::PubSubClient(const char*, uint16_t, std::function<void(char*, unsigned char*, unsigned int)>, Client&, Stream&)

    PubSubClient(const char*, uint16_t, MQTT_CALLBACK_SIGNATURE,Client& client, Stream&);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:116:4: note:   candidate expects 5 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:115:4: note: PubSubClient::PubSubClient(const char*, uint16_t, std::function<void(char*, unsigned char*, unsigned int)>, Client&)

    PubSubClient(const char*, uint16_t, MQTT_CALLBACK_SIGNATURE,Client& client);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:115:4: note:   candidate expects 4 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:114:4: note: PubSubClient::PubSubClient(const char*, uint16_t, Client&, Stream&)

    PubSubClient(const char*, uint16_t, Client& client, Stream&);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:114:4: note:   candidate expects 4 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:113:4: note: PubSubClient::PubSubClient(const char*, uint16_t, Client&)

    PubSubClient(const char*, uint16_t, Client& client);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:113:4: note:   no known conversion for argument 1 from 'WiFiClient' to 'const char*'

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:112:4: note: PubSubClient::PubSubClient(uint8_t*, uint16_t, std::function<void(char*, unsigned char*, unsigned int)>, Client&, Stream&)

    PubSubClient(uint8_t *, uint16_t, MQTT_CALLBACK_SIGNATURE,Client& client, Stream&);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:112:4: note:   candidate expects 5 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:111:4: note: PubSubClient::PubSubClient(uint8_t*, uint16_t, std::function<void(char*, unsigned char*, unsigned int)>, Client&)

    PubSubClient(uint8_t *, uint16_t, MQTT_CALLBACK_SIGNATURE,Client& client);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:111:4: note:   candidate expects 4 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:110:4: note: PubSubClient::PubSubClient(uint8_t*, uint16_t, Client&, Stream&)

    PubSubClient(uint8_t *, uint16_t, Client& client, Stream&);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:110:4: note:   candidate expects 4 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:109:4: note: PubSubClient::PubSubClient(uint8_t*, uint16_t, Client&)

    PubSubClient(uint8_t *, uint16_t, Client& client);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:109:4: note:   no known conversion for argument 1 from 'WiFiClient' to 'uint8_t* {aka unsigned char*}'

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:108:4: note: PubSubClient::PubSubClient(IPAddress, uint16_t, std::function<void(char*, unsigned char*, unsigned int)>, Client&, Stream&)

    PubSubClient(IPAddress, uint16_t, MQTT_CALLBACK_SIGNATURE,Client& client, Stream&);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:108:4: note:   candidate expects 5 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:107:4: note: PubSubClient::PubSubClient(IPAddress, uint16_t, std::function<void(char*, unsigned char*, unsigned int)>, Client&)

    PubSubClient(IPAddress, uint16_t, MQTT_CALLBACK_SIGNATURE,Client& client);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:107:4: note:   candidate expects 4 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:106:4: note: PubSubClient::PubSubClient(IPAddress, uint16_t, Client&, Stream&)

    PubSubClient(IPAddress, uint16_t, Client& client, Stream&);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:106:4: note:   candidate expects 4 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:105:4: note: PubSubClient::PubSubClient(IPAddress, uint16_t, Client&)

    PubSubClient(IPAddress, uint16_t, Client& client);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:105:4: note:   no known conversion for argument 1 from 'WiFiClient' to 'IPAddress'

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:104:4: note: PubSubClient::PubSubClient(Client&)

    PubSubClient(Client& client);

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:104:4: note:   candidate expects 1 argument, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:103:4: note: PubSubClient::PubSubClient()

    PubSubClient();

    ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:103:4: note:   candidate expects 0 arguments, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:83:7: note: PubSubClient::PubSubClient(const PubSubClient&)

 class PubSubClient {

       ^

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:83:7: note:   candidate expects 1 argument, 3 provided

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:83:7: note: PubSubClient::PubSubClient(PubSubClient&&)

C:\Users\Jonny\Documents\Arduino\libraries\PubSubClient\src/PubSubClient.h:83:7: note:   candidate expects 1 argument, 3 provided

ESPsonoff-v1.0p:38: error: 'MQTT' does not name a type

 void callback(const MQTT::Publish& pub) {

                     ^

ESPsonoff-v1.0p:38: error: expected unqualified-id before '&' token

 void callback(const MQTT::Publish& pub) {

                                  ^

ESPsonoff-v1.0p:38: error: expected ')' before '&' token

ESPsonoff-v1.0p:38: error: expected initializer before 'pub'

 void callback(const MQTT::Publish& pub) {

                                    ^

ESPsonoff-v1.0p:38: error: 'MQTT' does not name a type

 void callback(const MQTT::Publish& pub) {

                     ^

ESPsonoff-v1.0p:38: error: expected unqualified-id before '&' token

 void callback(const MQTT::Publish& pub) {

                                  ^

ESPsonoff-v1.0p:38: error: expected ')' before '&' token

ESPsonoff-v1.0p:38: error: expected initializer before 'pub'

 void callback(const MQTT::Publish& pub) {

                                    ^

exit status 1
no matching function for call to 'PubSubClient::PubSubClient(WiFiClient&, const char [14], int)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

KmanOz, 2017-04-07:
Are you using the right Pubsubclient that's I have specified in the instructions?

911061873, 2017-04-29:
在源码上面加入“#include <MQTT.h>”

KmanOz, 2017-04-30:
No the solution above it not the answer. You need to use the correct MQTT library that is specified in the README.

closed 2017-04-30 by KmanOz