KmanOz/Sonoff-HomeAssistant #19: Sometimes firmware yells "WiFi FAILED!" but device answers to pings.

linuxlurak, 2017-05-07:
HASS Version: 0.44
Sonoff Basic "2017 Version"

Repeatedly I saw the message "WiFi FAILED!" three or four times in a row in my arduino ide serial monitor on COM. I waited for the mqtt broker connection to show up with "ok". But nothing showed up.  Not even "Connecting to " appears. Instead the device restarts and prints again "WiFi FAILED!". 

I see in the sources that the devices restarts after failing to connect to wifi. So thats expected and working. No problem with restarting after failing. Problem seems to be the detection mechanism for successful wifi connection establishment.

At any time I can ping my device even if it thinks that wifi failed. There is no other device in my network with the same ip address. If I cut power to my Sonoff no pings return.

Any idea why connecting to wifi fails? Is the routine for checking the connection reliable/state of the art?

After some time wifi works obviously since I can switch the relay of my Sonoff.

What else can I provide you for debugging?

Cheers and thanks for your work.

1043717432, 2017-05-08:
I also have the same problem, the second day in the morning IP Ping barrier. Continuous attention, thank you, author

PonyOny, 2017-06-03:
I am having the same problem, seem it has hard time to connect to Wifi i'm receiving this several times and not conecting to Wifi, sometimes it connects just fine after some atempts.

> mqtt broker connection . . . . . . . . . . LOST
> 
>  ets Jan  8 2013,rst cause:1, boot mode:(3,7)
> 
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v3de0c112
> ~ld
> 
> 
> ----------------  Sonoff Powerpoint v1.01pOTA  -----------------
> 
> Unit ID: esp8266-A795F3
> Connecting to XXXXXX Wifi . . . . . . . . . . WiFi FAILED!
> 
> ----------------------------------------------------------------

Phileep, 2017-06-05:
I have had this issue also with my sonoff basics. I use my own firmware but they were cnstantly not identifying that they were connected to the wifi network.

This is an ESP8266 Arduino issue, not specific to this firmware.

It is noted in the [esp8266 arduino issues #2186](https://github.com/esp8266/Arduino/issues/2186) and forums 

I added to the setup section in my own code the following
```
  WiFi.persistent(false);
  WiFi.mode(WIFI_OFF); 
```
immediately before the line:
`  WiFi.mode(WIFI_STA);`

This has made my sonoffs stable with connecting to my wifi - suggest you try it also. It is a workaround until the arduino code is somehow fixed.

Try it in the kman code and see if it works for you.

1043717432, 2017-06-05:
@Phileep  Thank you for sharing, i am a novice, would i ask where to add in the firmware? Can I intercept an image for me?

Phileep, 2017-06-05:
1043717432, 

you have to add the 2 lines inside the relevant .ino file that you are using.

It should be in the section marked setup() - but you can search for the line `WiFi.mode(WIFI_STA);` and add the two lines suggested immediately above that line.

Then upload the code to the sonoff

1043717432, 2017-06-05:
@Phileep  Hi, is this added?
[image]

PonyOny, 2017-06-05:
Hi, thanks for reply, I added those lines of code but still facing the same issue :( that did not fix the problem, any other thing to try?

thanks.

Phileep, 2017-06-05:
sorry then, my only suggestion is read the issues thread for esp8266 in my link above - that is how I solved my issue.

It does seem to depend on what version of esp8266 arduino you are running etc.

linuxlurak, 2017-06-05:
@Phileep what is it about arduino? Do you mean the arduino IDE? That's the only thing about arduino. The esp-8266 (I am using an esp-01) has nothing else to do with arduino... The esp8266 is no arduino chip. It's a esp8266 chip. The modules just vombine some flash chip with the esp chip. So on the hardware side there ist nothing about arduino.

Phileep, 2017-06-05:
Tweaked my last comment to show I mean Arduino IDE. Again, see the link in my first post if you want more detail.

 My suggestion worked for me in my use case in my own Arduino sketch for the Sonoff.

Unfortunately, some peoples mileage has varied. 

That's all I have to contribute to this issue.

linuxlurak, 2017-06-05:
@Phileep: Thanks for the hint. I'll check on it as soon as  I get the time to.

KmanOz, 2017-06-11:
Thanks Phil. This is talked about in the Community Forum so didn't bother answering here. I'll leave this here for reference.