KmanOz/Sonoff-HomeAssistant #24: Hostname would be nice to have

donnib, 2017-06-08:
Hi,
When using OTA and when you have many devices it would be nice to be able to identify the nodes using a well known hostname, i know it's possible to use `WiFi.hostname("Name")` is this something you could consider adding ?

jgibbarduk, 2017-06-09:
[+1]
I also have a lot of Sonoff devices, it would be great to have their hostname show up in the Arduino IDE when wanting to update them OTA. Thanks.

KmanOz, 2017-06-09:
Yes this is something I am going to add as well as some diagnostics via mqtt. Stay tuned.

KmanOz, 2017-09-01:
Yeah I spent some time today getting the WiFi, OTA Update and MQTT username setup automatically by the ESP Chip ID. Works well and it something else you don't need to worry about. Anyway will keep at it and release another ver soon.

KmanOz, 2017-12-22:
this is now on the new version of the firmware

closed 2017-12-22 by KmanOz