KmanOz/Sonoff-HomeAssistant #28: Can't get the button to work on S20

donnib, 2017-06-27:
Hi,
I have flashed with 1.01pOTA and the switch works fine from HASS but i can't manually turn on / off from the button on the switch itself, have you experienced that ?

KmanOz, 2017-06-29:
Don't have one here too test but will have soon. I'll let you know

donnib, 2017-06-29:
Thank you, that would be cool to know, i would have guessed somebody else would have seen this problem if it was there as well known problem so either the HW is different or i did something wrong (which is hard to believe coz i didn't touch the code other than the configs.)

KmanOz, 2017-07-25:
Hey I have a question for you. I now have one here to test. How do you get them open LOL :D

donnib, 2017-07-25:
@KmanOz there is a screw underneath the small label on the back of it and then you pry it open.

mitsumaui, 2017-10-04:
Looks like this has been a while - however works fine for the S20 I have just flashed with the v1.01pOTA

closed 2017-12-22 by KmanOz