KmanOz/Sonoff-HomeAssistant #32: [REQUEST] Support for the new Sonoff RF Bridge 433

Gamester17, 2017-09-05:
Request firmware support for the new Sonoff RF Bridge 433

https://www.itead.cc/sonoff-rf-bridge-433.html

WiFi to 433MHz converter and vice versa to convert 433MHz RF sensors and remote to App via WiFi

https://www.itead.cc/wiki/Sonoff_RF_Bridge_433

Based on ESP8285 (ESP8266 chipset) from Itead.

https://www.youtube.com/watch?v=wTqQ_cmweuo

izaa, 2017-10-04:
@Gamester17 how do you know what chip that RF Bridge uses? I have Sonoff G1 and would like to also have a custom firmware but the chip is under fixes SIM-slot so cannot see what chip it uses.

Gamester17, 2017-10-05:
@izaa ESP8266 (or its sibling ESP8285) by Itead, and an EFM8BB1 by SiLabs for the RF radio.

Checkout this article => http://tinkerman.cat/hacking-sonoff-rf-bridge-433/

Note that both ESPurna firmware and the Sonoff-Tasmota are already supported on it now.