KmanOz/Sonoff-HomeAssistant #35: retained command messages overwrite rememberRelayState feature

skoslowski, 2017-09-24:
Thanks for sharing this. Perfect starting point for tinkering with sonoff devices!

While testing this I found a potential problem between the recommended settings for hass (publishing command messages with the retain flag set) and the EEPROM saved state. My original issue was that the switches would sometimes randomly change state for no apparent reason. Monitoring the mqtt messages I believe I found the reason:

Changing the relay state with the push button updates the state topic, the EEPROM, but not the retained command message value in the broker. So, if the device resets (say, after WIFI signal loss), it will first restore the saved state from the EEPROM, but then it will receive the retained message on its command topic, which overwrites that. This would mean, whenever the relay is switched-on remotely and turned-off locally (or vise versa) this results in somewhat expected behavior if the device ever looses its connection.

I am fairly new to MQTT and Arduino, but to my understanding the recommended hass settings should not include the retained flag. (Alternatively state toggle events from the push button on the device would need to publish the new state on the command topic)

KmanOz, 2017-09-25:
Actually has been an issue I'v known about for a very long time but just never got around to fixing it. v2 of the firmware will drop soon as I'm doing some final testing. Retain will be selectable as will QOS level. Glad someone picked it up :D

skoslowski, 2017-09-25:
Yeah, a note about this might have been helpful. But then I wouldn't have had any reason to do a deeper dive into this stuff, so all good here.

I can't seem to find a dev branch for v2? Any chance I can get a preview?

KmanOz, 2017-09-25:
That's cause I haven't created it yet. In the next couple of days I will post something up. I'll keep you posted.